alexdelprete/ha-sinapsi-alfa #12: Login error

ninoger, 2024-04-02:
### System Health details

## System Information

version | core-2024.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.12.2
os_name | Linux
os_version | 6.6.20-haos
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1417
Downloaded Repositories | 56

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 37

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 12.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.03.1
agent_version | 1.6.0
docker_version | 24.0.7
disk_total | 468.7 GB
disk_used | 28.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Samba share (12.3.1), Log Viewer (0.17.0), Duck DNS (1.16.0), NGINX Home Assistant SSL proxy (3.9.0), FTP (5.0.2), Home Assistant Google Drive Backup (0.112.1), ESPHome (2024.3.1), Mosquitto broker (6.4.0), Zigbee2MQTT (1.36.1-1), Studio Code Server (5.15.0), Piper (1.5.0), openWakeWord (1.10.0), ZigStar Silicon Labs FW Flasher (0.1.1), ZigStar TI CC2652P/P7 FW Flasher (0.4.0), Advanced SSH & Web Terminal (17.2.0), AdGuard Home (5.0.5), Assist Microphone (1.2.0), Matter Server (5.5.0), Whisper (2.0.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 37
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 28 marzo 2024 alle ore 16:58
-- | --
current_recorder_run | 2 aprile 2024 alle ore 22:56
estimated_db_size | 2675.73 MiB
database_engine | sqlite
database_version | 3.44.2

</details>

<details><summary>Sonoff</summary>

version | 3.6.0 (8dd8af9)
-- | --
cloud_online | 24 / 31
local_online | 16 / 16

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Quando immetto i dati per l'accesso (IP), ottengo sempre: Unknown error occurred.
Ho eliminato anche l'integrazione manuale dal configuration.yaml, riavviato, ma non è cambiato nulla. Sempre lo stesso errore di accesso. Grazie

### Reproduction steps

1. inserisco IP
2. Invia
3. Unknown error occurred
...

### Debug logs

```text
File "/config/custom_components/sinapsi_alfa/api.py", line 276, in read_modbus_alfa
    read_data.registers, byteorder=Endian.BIG
 File "/config/custom_components/sinapsi_alfa/config_flow.py", line 91, in async_step_user
    uid = await self.get_unique_id(name, host, port, scan_interval)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/config_flow.py", line 66, in get_unique_id
    self.api_data = await self.api.async_get_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/api.py", line 220, in async_get_data
    result = await self._hass.async_add_executor_job(self.read_modbus_alfa)
```

alexdelprete, 2024-04-03:
Verifica che raggiungi IP dalla shell del container HA (installa addon SSH), usando il comando "ping IP".

Possibilmente, fornisci l'intero log debug.

ninoger, 2024-04-03:
Ciao, spero di riuscire a mettere il log completo (non ho molta dimestichezza). Comunque ti anticipo che l'integrazione manuale funziona. Ho tutti i sensori previsti dal codice fornito da Alfa. Quindi HA e il device si parlano regolarmente. Grazie.

alexdelprete, 2024-04-03:
non puoi avere l'integrazione alfa insieme ai sensori configurati manualmente, devi rimuovere i sensori creati manualmente. ho creato l'integrazione proprio per non usare la configurazione modbus, se usi entrambi avrai HA che usa l'integrazione modbus e crea i sensori manualmente e la mia integrazione Alfa che fa la stessa cosa ma crea tutti i sensori automaticamente. Inutile duplicazione e probabilmente le due vanno in conflitto.

ninoger, 2024-04-03:
Ho appena provato il ping e risponde

alexdelprete, 2024-04-03:
Elimina i sensori creati manualmente e riavvia HA.

alexdelprete, 2024-04-03:
Mandami il full debug log, altrimenti non so come aiutarti.

ninoger, 2024-04-03:
Spero di riuscire. Ormai domani. Farò pulizia completa e riprovo, sperando di risolvere e di ricavare il log completo. Intanto grazie. Buona notte

alexdelprete, 2024-04-03:
Per caso usi VLAN nella tua rete e magari l'Alfa è in una VLAN diversa da HA?

ninoger, 2024-04-03:
No, niente. Ho un semplice Fritzbox 7530 con reti 2.4 e 5 ghz separate e alcuni repeater in mesh, sempre fritz.

alexdelprete, 2024-04-03:
Anche io uso fritz 7590 e repeater 3000 e 2400. Unica mesh, non separata.

Va bene, fammi sapere.

ninoger, 2024-04-03:
Ho dimenticato di dirti che, in fase di scarico dell'integrazione da hacs, visualizza la schermata introduttiva di un'altra tua integrazione, anche se poi scarica il pacchetto giusto.

alexdelprete, 2024-04-03:
> Ho dimenticato di dirti che, in fase di scarico dell'integrazione da hacs, visualizza la schermata introduttiva di un'altra tua integrazione, anche se poi scarica il pacchetto giusto.

??? mi fai vedere uno screenshot?

alexdelprete, 2024-04-03:
azz...hai ragione...ora sistemo, grazie per la segnalazione. :)

ninoger, 2024-04-03:
[image]

E ' la stessa schermata che ti viene fuori da te?

alexdelprete, 2024-04-03:
Sì ma non riesco a trovare dove peschi quel readme...quello su questo repository è corretto. Mah...

ninoger, 2024-04-14:
Ciao, scusami se sono scomparso, ma sono stato assente. Ancora non riesco a loggarmi con la tua integrazione. Ho messo su una nuova istanza di HA Core, aggiungendo solo HACS, su un Sonoff iHost. Anche li non riesco a loggarmi ricevendo lo stesso identico errore. Ti allego il log completo di Core di questa istanza aggiornata e senza altre integrazioni presenti (non ho configurato nulla). Non so se ti possono servire altri tipi di log. Grazie.
[home-assistant_2024-04-14T00-44-58.645Z.log](https://github.com/alexdelprete/ha-sinapsi-alfa/files/14969080/home-assistant_2024-04-14T00-44-58.645Z.log)

alexdelprete, 2024-04-14:
Vedo errori di base modbus, e dal log non mi sembra una installazione pulita, vedo altri componenti installati etc.

Sembra che modbus non stia funzionando proprio nella tua istanza. Sei sicuro l'ip sia giusto? Prova a disinstallare il mio componente e configurarlo col modbus file statico come da istruzioni di sinapsi, e vedi se funzionano i sensori.

ninoger, 2024-04-14:
Alfa è con ip statico e sta funzionando regolarmente con la sua app. Il log è di una istanza pulita con solo hacs e la tua integrazione. Ovviamente mi rileva molti device ma in questa istanza non li ho configurati, ovviamente. Nell'altra istanza, quella dell'altra volta, col codice di Alfa nel configuration.yaml i sensori ci sono e funziona. Per fare quest'ultima prova ho spento il mini pc per evitare conflitti. È ovvio che ho utilizzato lo stesso ip statico che ho nell'app Alfa (192.168.1.140).

alexdelprete, 2024-04-14:
Quindi Alfa sta già rispondendo via modbus alla integrazione manuale sull'altra istanza.

Elimina TUTTI i sw che comunicano con Alfa. Compresa quella integrazione manuale e riconfigurare solo la mia integrazione.

ninoger, 2024-04-14:
Quindi anche l'app Alfa? Per quanto riguarda l'altra istanza, avevo già commentato tutte le righe dell'integrazione manuale. Ma aveva dato lo stesso errore. Per provare con questa nuova istanza, ho pure spento il mini pc che utilizzo con haos. Comunque, rifaccio tutto. Grazie.

PS: continua a visualizzare l'altra tua integrazione quando si scarica da hacs.

alexdelprete, 2024-04-14:
No, l'app Alfa no redo comunichi via modbus.

Domanda, che versione di HA stai usando? Quell'errore su modbus sembra di base, mica stai usando roba vecchia?

ninoger, 2024-04-14:
No, tutto all'ultima versione. Aggiorno sempre tutto subito.

alexdelprete, 2024-04-14:
Altra cosa, l'errore del primo post non lo vedo nel log. Non riesco quindi a capire quale errore tu abbia ora. Dimmi che versione del componente stai usando e che versione di HA. Abilita il debug e ridammi il log dopo aver riprodotto l'errore.

ninoger, 2024-04-14:
Il log parziale del primo post è quello del mini pc con haos, ed è l'istanza di "produzione". Il log completo che ho allegato è dell'istanza di prova che ho messo su Sonoff iHost (è Core/docker). La versione è 2024.4.3.
Come faccio ad abilitare il debug? Grazie

alexdelprete, 2024-04-14:
Entra nell'integrazione e in basso a sx sotto il logo trovi debug come ultima voce.

ninoger, 2024-04-14:
Scusa, come faccio a entrare nell'integrazione se non mi consente la configurazione? Vedi screenshot dell'altra volta

ninoger, 2024-04-14:
Finalmente è andata sulla nuova istanza e  con l'altra istanza (mini pc) spenta completamente. Ora vedo 27 entità e sono funzionanti. Ho riacceso il mini pc e, come mi aspettavo, Alfa funziona solo nella prima istanza che ha l'integrazione manuale. Ma la tua integrazione potrebbe convivere nelle due istanze, eliminando, ovviamente, l'integrazione manuale dal mini pc? Grazie per il supporto.

alexdelprete, 2024-04-14:
Allora, ti avevo già detto che il device non credo supporti più di una sessione modbus per volta. Quindi devi interrompere qualsiasi altra sessione se vuoi usare la mia integrazione.

Non riesco a capire cosa tu abbia configurato, ma qualsiasi cosa sia ELIMINALA, ed usa soltanto la mia integrazione. Oppure cancella la mia integrazione e usa il file fornito da sinapsi con i registri modbus manuali.

Di più non so come aiutarti.

ninoger, 2024-04-14:
Boh, mi sembra che ti sei irritato. Se è così mi dispiace, ma ho cercato di spiegare cosa avevo fatto e cosa ho fatto. E alla fine ha funzionato. Ho la sensazione che in molti, quando uno chiede spiegazioni, magari sbagliando, alla fine non leggete i quesiti. A me sembra di essere stato cortese e modesto. Ad ogni modo ti ringrazio per il lavoro che condividi con tutti.

alexdelprete, 2024-04-14:
Non mi sono irritato, tranquillo. è solo che se ti dico che devi interrompere le sessioni verso il device e ora mi dici che funziona perché hai spento l'altro pc, significa che sull'altro pc c'era qualcosa che accedeva all'alfa via modbus. :)

alexdelprete, 2024-04-14:
Se ora sei a posto per cortesia chiudi l'issue.

closed 2024-04-14 by ninoger

dvbit, 2024-05-21:
Grazie. per il thread . avevo lo stesso problema. e grazie per l'integrazione