alexdelprete/ha-sinapsi-alfa #19: K=25

stepsolar, 2024-04-25:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hi, I have another alpha installed on a three-phase meter with 80kw power. From alpha they had to modify the reading values ​​per app by multiplying by 25 so as to have the real values, but I notice that these multiplied values ​​on the alpha app are not read on your integration which still reads the non-multiplied values. Would it be possible to multiply by 25? It would be nice to include in the configuration both the alpha IP and another field whether or not to multiply by 25 which would be the K coefficient. K=25 is also talked about on the internet. Thank you

### Describe the solution you'd like

.

### Describe alternatives you've considered

.

### Additional context

.

alexdelprete, 2024-04-25:
Ask Sinapsi to publish these values in modbus. Probaly the register map changed and they added these specific values in other addresses. I have to stick to the default register map.

Meanwhile, if it’s just a *25 problem, you can create template sensors and multiply the alpha sensors by 25.

stepsolar, 2024-04-25:
Thanks, I will ask sinaspi if this modification is possible.  I had also thought about creating other sensors *25 but I don't like it.  Thank you

alexdelprete, 2024-04-25:
What is this K coefficient? I can't find much about K=25, you said there is docs on the internet.

alexdelprete, 2024-04-25:
Ok, found it. Sinapsi has to publish the K coefficient in modbus, so I will always multiply it, default should be 1, so it's ok, if in your case it's 25 it will show the correct values.

[image]

stepsolar, 2024-04-25:
On their app they have already changed the value but on the modbus I still receive the non-multiplied values

alexdelprete, 2024-04-25:
> On their app they have already changed the value but on the modbus I still receive the non-multiplied values

Ask them if the Power Factor is available in the modbus registers. If not, ask them to implement it. ;)

I asked them to implement the serial number of the device in modbus, but they didn't do it.

stepsolar, 2024-04-25:
I have already sent an email, I hope now that they will also modify the K coefficient in Modbus, otherwise I will create the sensors by multiplying them by 25. Couldn't you create the possibility of multiplying by 25 in your app?  So do you automatically create the correct sensors?

alexdelprete, 2024-04-25:
You don't like template sensors, and I don't like managing the K factor manually in the integration, it should be provided by the meter, as Shelly EM or other meters do.

In case Sinapsi won't implement it, I'll think about it...but I don't like it.

alexdelprete, 2024-04-25:
> Thanks, if you increase it I think it will be useful for me but also for others, otherwise you always have to ask sinaspi since it is an unmanageable setting.

you forget 1 thing: in your case K=25 but in other cases it might be a different value that is also dynamic, not static, so it can change. Having it configured as a static value is a wrong approach.

K must be provided by the meter.

alexdelprete, 2024-04-25:
Te lo spiego in italiano: nel tuo caso il K=25, ma in altri casi il K potrebbe variare dinamicamente. è errato metterlo come valore statico.

mi spiace ma non lo farò. creati dei template sensor per ora, e cerca di spiegare a Sinapsi che il K deve essere disponibile nella mappa modbus.

stepsolar, 2024-04-30:
Mi hanno risposto, non riescono a moltiplicare i codici modbus x25. La soluzione è creare dei nuovi sensori moltiplicati. Che brutta cosa..

alexdelprete, 2024-04-30:
Non riescono? 😂

Non vogliono, è diverso.

stepsolar, 2024-05-01:
Ho notato che alcuni valori sono corretti come ad esempio potenza prodotta, la potenza immersa deve essere moltiplicata e automaticamente il valore potenza auto consumata è sbagliata perché viene soltratto con un valore errato..

alexdelprete, 2024-05-01:
Fammi vedere schermata con i sensori, qui i calcoli sembrano corretti. Mostrami schermata con tutti i sensori e descrivimi perché non ti tornano i calcoli.

alexdelprete, 2024-05-01:
Apri un altro issue con le info che ti ho richiesto, questo è chiuso.

stepsolar, 2024-05-01:
Allego foto con le differenze... 
[image]
[image]

alexdelprete, 2024-05-01:
L'unico valore calcolato è l'autoconsumo, il resto me lo restituisce Alfa dai registri modbus.

Quale valore non ti torna, l'autoconsumo?

stepsolar, 2024-05-01:
Si,
Autoconsumo non torna.

alexdelprete, 2024-05-01:
Dai tuoi screenshot non si vede Energia immessa totale.

Il calcolo è semplice: `energia/potenza autoconsumata = energia/potenza prodotta - energia/potenza immessa`

Questo è il codice:

```python
    if reg_type == "calcolato":
        self.data["potenza_auto_consumata"] = (
            self.data["potenza_prodotta"] - self.data["potenza_immessa"]
        )
        self.data["potenza_consumata"] = (
            self.data["potenza_auto_consumata"]
            + self.data["potenza_prelevata"]
        )

        self.data["energia_auto_consumata"] = (
            self.data["energia_prodotta"] - self.data["energia_immessa"]
        )
        self.data["energia_consumata"] = (
            self.data["energia_auto_consumata"]
            + self.data["energia_prelevata"]
        )
```

stepsolar, 2024-05-02:
Dovrei sostituire i tuoi sensori con i miei che ho dovuto moltiplicare x 25.
Altrimenti se ci fosse la possibilità al tuo codice inserirei la moltiplicazione così poi le vari sostrazzioni saranno con i valori corretti. Dove potrei inserire la moltiplicazione?

alexdelprete, 2024-05-02:
Ma che ci vuole a creare dei template sensor dove ti fai la moltiplicazione. Non capisco cos'è che ti dia fastidio...:)

stepsolar, 2024-05-03:
Si li ho ho creati ma non mi piace, quando apro la tua integrazione vedo tutti i dati sbagliati e questo non lo sopporto. Ho atteso tanto per questa integrazione e adesso..

alexdelprete, 2024-05-03:
Devi prendertela con Sinapsi. Non ci vuole molto a pubblicare il K nei registri modbus. Se lo fanno, implementerò la modifica.

stepsolar, 2024-05-05:
Mi hanno risposto "noi a monte non fissiamo fare niente, deve configurare homeassistant in modo che i valori vengono moltiplicati per 25"
Che tristezza... Ti devi immaginare che ho notato di nuovo valori non corretti sulla loro app come se il parametro 25 sia stato tolto, e ho fatto nuovamente segnalazione.. Spero che non danno adesso la colpa al distributore "E-distribuzione"...

alexdelprete, 2024-05-05:
Digli così: "il fattore K viene fornito dal contatore e-distribuzione, quindi Alfa dovrebbe leggerlo e calcolare i valori REALI e fornire valori REALI nell'app e via modbus. Dovrebbe anche fornire via modbus il fattore K letto dal contatore."

stepsolar, 2024-05-06:
Ecco qui la risposta che mi hanno dato: "Buongiorno,
non è così perché i valori forniti dal contatore sono privi della costante di trasformazione K, per questo motivo devono essere sempre moltiplicati per quel valore."

alexdelprete, 2024-05-06:
Dalla documentazione del contatore openmeter si evince che il valore viene mostrato dal contatore:

[image]

stepsolar, 2024-05-07:
Sono andato nel contatore, cliccando sul display non ho trovato nulla sul valore K o valore 25.

alexdelprete, 2024-05-07:
> Sono andato nel contatore, cliccando sul display non ho trovato nulla sul valore K o valore 25.

La documentazione dice che lo trovi nel menu "Lista spiegazione simboli". Ma io non ho quel contatore, chiedi ad e-distribuzione come visualizzarlo.

closed 2024-12-01 by alexdelprete