alexdelprete/ha-sinapsi-alfa #74: Polling interval < 30s

sigitm, 2024-12-01:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

"Potenza prelevata" sensor is not updated in realtime.

### Describe the solution you'd like

I'd like to choose a smaller polling interval, e.g. 1 or 2 seconds.

### Describe alternatives you've considered

It worked with Modbus integration

### Additional context

It's about two years since I've been using ALFA with "standard" Modbus integration.
I declared "scan_interval: 1" to the first sensor "ALFA Potenza Attiva Prelevata Istantanea" and it works like a charm.
Whenever I switch on too many appliances, the sensor updates almost instantaneously, and I can take proactive measures (e.g. with notifications). 

Is it possible to integrate this feature?

At least for the "Potenza prelevata" sensor.

Thanks.

alexdelprete, 2024-12-01:
The polling frequency is the same for the single "coordinator" (the entity responsible to coordinate data fetching).

I don't see ANY rational use-case for polling at that frequency, also because the meter does not send data so frequently, it only changes when there's a sensible change in data.

[image]

alexdelprete, 2024-12-01:
Nothing stops you from disabling the "Potenza prelevata" sensor of this integration, and configuring it as a standard modbus sensor, like you did before.

closed 2024-12-01 by alexdelprete

sigitm, 2026-01-07:
Hi @Cippo2019 
I use direct modbus integration as specified here: https://www.alfabysinapsi.it/home-assistant/

On some entities where I need near-real time updates, according to "modbus integration documentation ( https://www.home-assistant.io/integrations/modbus/ ) I added the following attribute:

**scan_interval: 2**

so the information is updated every 2 seconds.

For example:

```
    sensors:
        - name: ALFA Potenza Attiva Prelevata Istantanea
          device_class: power
          unit_of_measurement: W
          state_class: measurement
          data_type: uint16
          address: 2
          scan_interval: 2
```

This works perfectly.
For example, if I turn on the electric oven in my kitchen, after about 2 seconds I see the updated consumption in Home Assistant. Therefore I can implement all sorts of notifications.

I think it’s very useful to have this information in real time. As you can see in the thread, I asked @alexdelprete to implement this in this integration as well, but he was not willing to do it.

Cippo2019, 2026-01-07:
hi and thanks to all for the clarification
one point
I am using modbus to query my solaredge inverter every 3 seconds (is under testing in one test HA, not in production)
to be honest I dont see any alert/warning in home assistant..... ,,,,mmmm

alexdelprete, 2026-01-08:
> I am using modbus to query my solaredge inverter every 3 seconds (is under testing in one test HA, not in production)
> to be honest I dont see any alert/warning in home assistant..... ,,,,mmmm

i showed you the source code of HA Modbus integration: if poll_interval is < 5s you get a warning in the log. Period. It's not an opinion. :)

> any possibility from Alfa to "know" how we can use mqtt or rest ?

They don't want us to use them. Who knows why...

Cippo2019, 2026-01-09:
> i showed you the source code of HA Modbus integration: if poll_interval is < 5s you get a warning in the log. Period. It's not an opinion. :)
>
>
> They don't want us to use them. Who knows why...

i saw only now your note.. I am beleving you but I am asking to myself why I dont see that alert ....