alexdelprete/ha-sinapsi-alfa #92: Upgrade Pymodbus to 3.8.0

Aljosha89, 2025-01-31:
### System Health details

Impossible to load integration

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

According to huawei solar integration, all the integrations that are using pymodbus should upgrade to pymodbus 3.8.0 (included in HA Core 2025.2)

As a matter of fact on HA Core 2025.2.0 beta2 Huawei Solar integration work, but not this one (and ABB integration neither)

More detail on this link: https://github.com/wlcrs/huawei_solar/issues/873#issuecomment-2626681291)

### Reproduction steps

Simply upload to 2025.2.0 with other integrations that are using Modbus

### Debug logs

```text
None
```

alexdelprete, 2025-02-02:
All custom integrations should support HA system libraries versions: for HA 2025.1.4 (latest official) it's 3.7.4: https://github.com/home-assistant/core/blob/6145ea232367ad2364b40eddd687a11943298803/requirements_all.txt#L2096

For HA 2025.2.0b6 it's 3.8.3: https://github.com/home-assistant/core/blob/dc1c2f24e62f31030540be22a87714e02ced5bac/requirements_all.txt#L2121

In my integration I map the latest minimum official requirement (3.7.4), but if pymodbus greater than 3.7.4 is loaded, it should work anyway, see here: https://github.com/alexdelprete/ha-sinapsi-alfa/blob/f1d93efddc03f6ccf6adc76f423da500942f3b60/custom_components/sinapsi_alfa/manifest.json#L11C3-L11C56

```python
"requirements": ["pymodbus>=3.7.4", "getmac>=0.9.4"],
```python

This means that the integration requires _**at least**_ pymodbus 3.7.4. So it should load also in HA 2025.2.x if pymodbus 3.8.x is loaded.

If it doesn't work, you need to provide detailed debug logs so that I can check what's going on.

Aljosha89, 2025-02-02:
Tried to update now to 2025.2.0b6 but I cannot load the integration. Not sure If I've taken the log correctly, in the meantime I reverted to previous version:

`
2025-02-02 19:29:09.451 INFO (MainThread) [custom_components.sinapsi_alfa] 
-------------------------------------------------------------------
Sinapsi Alfa
by @alexdelprete
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/alexdelprete/ha-sinapsi-alfa/issues
-------------------------------------------------------------------

2025-02-02 19:29:09.451 DEBUG (MainThread) [custom_components.sinapsi_alfa] Setup config_entry for sinapsi_alfa
2025-02-02 19:29:09.451 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
2025-02-02 19:29:09.451 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.83.15:502 with a 3.0s timeout.
2025-02-02 19:29:09.593 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:29:09.594 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Get_Mac_Address (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Get_Mac_Address (SUCCESS): found mac address 083AF23C4AAC
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Coordinator Config Data: {'host': '192.168.83.15', 'name': 'Alfa', 'port': 502, 'scan_interval': 60.0}
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Coordinator init - Host: 192.168.83.15 Port: 502 ScanInterval: 60
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Data Coordinator: Update started at 2025-02-02 19:29:09.597460
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] API Client connect to IP: 192.168.83.15 port: 502 timeout: 59
2025-02-02 19:29:09.597 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.83.15:502 with a 3.0s timeout.
2025-02-02 19:29:09.610 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:29:09.610 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Inverter ready for Modbus TCP connection
2025-02-02 19:29:09.625 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Modbus TCP Client connected
2025-02-02 19:29:09.625 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Start Get data (Host: 192.168.83.15 - Port: 502)
2025-02-02 19:29:09.625 DEBUG (SyncWorker_2) [custom_components.sinapsi_alfa.api] (read_modbus_alfa) Key: potenza_prelevata Addr: 2 Type: uint16 DevClass: power
2025-02-02 19:29:09.625 DEBUG (SyncWorker_2) [custom_components.sinapsi_alfa.api] (read_modbus_alfa): failed with error: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given

2025-02-02 19:30:10.484 DEBUG (MainThread) [custom_components.sinapsi_alfa] Setup config_entry for sinapsi_alfa
2025-02-02 19:30:10.484 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
2025-02-02 19:30:10.485 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.83.15:502 with a 3.0s timeout.
2025-02-02 19:30:10.518 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:30:10.518 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Get_Mac_Address (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Get_Mac_Address (SUCCESS): found mac address 083AF23C4AAC
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Coordinator Config Data: {'host': '192.168.83.15', 'name': 'Alfa', 'port': 502, 'scan_interval': 60.0}
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Coordinator init - Host: 192.168.83.15 Port: 502 ScanInterval: 60
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Data Coordinator: Update started at 2025-02-02 19:30:10.519271
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] API Client connect to IP: 192.168.83.15 port: 502 timeout: 59
2025-02-02 19:30:10.519 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.83.15:502 with a 3.0s timeout.
2025-02-02 19:30:10.528 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.83.15:502
2025-02-02 19:30:10.529 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Inverter ready for Modbus TCP connection
2025-02-02 19:30:10.547 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Modbus TCP Client connected
2025-02-02 19:30:10.547 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Start Get data (Host: 192.168.83.15 - Port: 502)
2025-02-02 19:30:10.548 DEBUG (SyncWorker_0) [custom_components.sinapsi_alfa.api] (read_modbus_alfa) Key: potenza_prelevata Addr: 2 Type: uint16 DevClass: power
2025-02-02 19:30:10.548 DEBUG (SyncWorker_0) [custom_components.sinapsi_alfa.api] (read_modbus_alfa): failed with error: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given
2025-02-02 19:30:10.548 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Coordinator Update Error:  at 2025-02-02 19:30:10.484984
2025-02-02 19:30:10.548 DEBUG (MainThread) [custom_components.sinapsi_alfa.coordinator] Finished fetching sinapsi_alfa (083AF23C4AAC) data in 0.030 seconds (success: False)
`

alexdelprete, 2025-02-02:
> ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given

I have to check if pymodbus 3.8.3 changed read_holding_registers(), in any case, I am not supporting HA beta versions. Install official releases, those are tested.

Aljosha89, 2025-02-02:
I agree, I was just informing in advance since official release is three days away ;-)

Thanks
Aljosha

alexdelprete, 2025-02-02:
> I agree, I was just informing in advance since official release is three days away ;-)
> 
> Thanks Aljosha

did you test it with HA 2025.1.x?

alexdelprete, 2025-02-02:
> did you test it with HA 2025.1.x?

it's strange: read_holding_registers() didn't change in pymodbus 3.8.x, so it should've worked.

I just made a small adjustment, can you please test latest `0.3.0-beta.1` and provide the debug log? If it works, I'll release 0.3.0 final.

Thanks.

Aljosha89, 2025-02-02:
Unfortunately I did not tried with 2025.1 because Huawei inverter integration was broken with 2025.1 and I rely on that integration for most of my smart home automation. 

Same error with 0.3.0 beta 1:

`2025-02-02 22:44:49.955 DEBUG (SyncWorker_5) [custom_components.sinapsi_alfa.api] (read_modbus_alfa): failed with error: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given`

alexdelprete, 2025-02-02:
Ok, found the issue:

[image]

pymodbus has this bad habit of changing very often the API, and this leads to tons of issue for integration developers. I'll release beta.2 so you can test it.

Aljosha89, 2025-02-03:
Beta 2 works like a charm! 

> [@Aljosha89](https://github.com/Aljosha89) please test beta.2 log and let me know.

closed 2025-02-03 by alexdelprete