alexdelprete/ha-sinapsi-alfa #117: Energia Prodotto non conforme

Valentes80, 2025-05-10:
### System Health details

nessun problema riscontrato

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Complimenti per l'integrazione davvero funzionale e meno macchinosa del dover installare su yaml tutto il codice disponibile sul sito sinapsi. Noto però che i sensori :  Energia Prodotta, Potenza Auto Consumata, Potenza Consumata, Potenza Immessa , Potenza Immessa Media , e Potenza Prodotta, mi danno corrispondenza 0 anche quando l'impianto fotovoltaico è in produzione e comunque dell'applicazione sinapsi viene correttamente invece vista.
se posso darti supporto in qualche modo, sono a completa disposizione.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
nessun bug se non quello descritto nel messaggio
```

alexdelprete, 2025-05-10:
Praticamente non ti funziona l'integrazione quindi? Tutti i valori a zero? Fammi vedere tutti i sensori e i valori.

Dopodiché, abilita il debug log, fallo girare per qualche minuto, e poi scaricalo e postalo qui, così vediamo cosa succede.

Valentes80, 2025-05-10:
[image] no alcuni sensori funzionano tipo potenza prelevata ecc... Gli altri danno valori che non hanno senso e sopratutto quali che a me serve tipo energia prodotta o potenza prodotta mi da 0

alexdelprete, 2025-05-10:
Hai abilitato il debug log sull'integrazione come ti ho chiesto nel post precedente? Devo verificare cosa succede.

Nel frattempo, fai anche un'altra prova: disabilita l'integrazione e usa lo yaml ufficiale, vedi se i sensori riportano valori corretti.

Valentes80, 2025-05-11:
Non so come si fa... Ho provato ad applicare i sensori tramite yaml ma effettivamente anche i sensori che ne escono danno valori animali rispetto all'applicazione

alexdelprete, 2025-05-12:
disabilita l'integrazione. lascia solo i sensori tramite yaml, e contatta il supporto alfa dicendogli che i valori dei sensori non sono corretti.

closed 2025-06-16 by alexdelprete

matteoopc, 2025-07-21:
Riapro l'argomento perchè ho lo stesso problema...l'unico valore giusto è la potenza utilizzata in tempo reale. Idee?

alexdelprete, 2025-07-22:
l'issue è chiuso, non si commenta un issue chiuso, non lo puoi riaprire.

se vuoi supporto, apri un nuovo issue, fornisci un debug log dove si vedono i valori dei sensori e forse saprò dirti qualcosa. non servono idee, servono DATI per capire cosa sta rispondendo il tuo Alfa.