alexdelprete/ha-sinapsi-alfa #130: L'integrazione non si carica con HA Core 2025.7.0

ninoger, 2025-07-02:
### System Health details

Dopo aver aggiornato HAOS all'ultima 2025.7.0 l'integrazione non si carica più. Fino alla 2025.6.3 tutto ok.

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Non riesco a leggere log in quanto sembra che sia stata eliminata

### Reproduction steps

1.Non carica affatto
2.
3.
...

### Debug logs

```text
Non so come ricavare log
```

Donez84, 2025-07-02:
Hi, same issue unfortunatly. 

[home-assistant_2025-07-02T21-38-00.504Z.log](https://github.com/user-attachments/files/21025963/home-assistant_2025-07-02T21-38-00.504Z.log)

alexmil72, 2025-07-03:
The error is:
Setup failed for custom integration 'sinapsi_alfa': Unable to import component: cannot import name 'pack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)

`ImportError: cannot import name 'pack_bitstring' from 'pymodbus.utilities'`

The custom component sinapsi_alfa is trying to import a function (pack_bitstring) that no longer exists in the new version of the pymodbus library included in HA.

Sk1zZo, 2025-07-03:
Same here

Setup failed for custom integration 'sinapsi_alfa': Unable to import component: cannot import name 'pack_bitstring' from 'pymodbus.utilities' (/usr/local/lib/python3.13/site-packages/pymodbus/utilities.py)

alexdelprete, 2025-07-03:
Update to latest version: 0.3.2.

alexmil72, 2025-07-03:
> Update to latest version: **0.3.2**

It works, many thanks!

closed 2025-07-03 by alexdelprete