alexdelprete/ha-sinapsi-alfa #136: Error when configuring Alpha integration

JavaJix, 2025-07-27:
### System Health details

Home Assistant Core on docker v. 2025.7.3

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Configuring the integretion gives an "unknown error message" and i found this on homeassistant.log file:

File "/config/custom_components/sinapsi_alfa/config_flow.py", line 96, in async_step_user
    uid = await self.get_unique_id(name, host, port, scan_interval)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/config_flow.py", line 69, in get_unique_id
    self.api = SinapsiAlfaAPI(self.hass, name, host, port, scan_interval)
               ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/api.py", line 104, in __init__
    raise ValueError("Scan interval must be at least 10 seconds")

### Reproduction steps

1. configure home assistant integration
2.
3.
...

### Debug logs

```text
File "/config/custom_components/sinapsi_alfa/config_flow.py", line 96, in async_step_user
    uid = await self.get_unique_id(name, host, port, scan_interval)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/config_flow.py", line 69, in get_unique_id
    self.api = SinapsiAlfaAPI(self.hass, name, host, port, scan_interval)
               ~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/api.py", line 104, in __init__
    raise ValueError("Scan interval must be at least 10 seconds")
```

JavaJix, 2025-07-27:
[image]

alexdelprete, 2025-07-29:
Please test [v0.4.1-beta.1](https://github.com/alexdelprete/ha-sinapsi-alfa/releases/tag/v0.4.1-beta.1) and let me know if it works.

Thanks for reporting the problem.

alexdelprete, 2025-07-30:
Try [beta 3](https://github.com/alexdelprete/ha-sinapsi-alfa/releases/tag/v0.4.1-beta.3) and let me know.

alexdelprete, 2025-08-02:
In order to enable debug before installing the integration, use logger config:

```yaml
logger:
  default: warning
  logs:
    custom_components.sinapsi_alfa: debug
```

JavaJix, 2025-08-03:
Debug enabled, below you will find the logs:

----

2025-08-03 08:04:23.913 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] Test connection to 192.168.10.166:502
2025-08-03 08:04:23.913 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] Creating API Client
2025-08-03 08:04:23.913 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] API Client created: calling get data
2025-08-03 08:04:23.913 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] API Client connect to IP: 192.168.10.166 port: 502 timeout: 59.0
2025-08-03 08:04:23.913 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-03 08:04:23.965 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-03 08:04:23.965 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Inverter ready for Modbus TCP connection
2025-08-03 08:04:24.071 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Modbus TCP Client connected
2025-08-03 08:04:24.071 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-03 08:04:24.168 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-03 08:04:24.168 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 1: SUCCESS
2025-08-03 08:04:24.169 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 1 failed, retrying in 2.0s
2025-08-03 08:04:26.135 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 2 failed, retrying in 2.1s
2025-08-03 08:04:28.205 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 3 failed, retrying in 4.9s
2025-08-03 08:04:33.105 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-03 08:04:33.181 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-03 08:04:33.182 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 4: SUCCESS
2025-08-03 08:04:33.182 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 4 failed, retrying in 9.0s
2025-08-03 08:04:42.134 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 5 failed, retrying in 10.0s
2025-08-03 08:04:52.136 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 6 failed, retrying in 10.0s
2025-08-03 08:05:02.137 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-03 08:05:02.261 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-03 08:05:02.262 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 7: SUCCESS
2025-08-03 08:05:02.262 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 7 failed, retrying in 10.0s
2025-08-03 08:05:12.262 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 8 failed, retrying in 10.0s
2025-08-03 08:05:22.265 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 9 failed, retrying in 10.0s
2025-08-03 08:05:32.265 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-03 08:05:32.369 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-03 08:05:32.369 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 10: SUCCESS
2025-08-03 08:05:32.370 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC address not found after all attempts
2025-08-03 08:05:32.370 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Start Get data (Host: 192.168.10.166 - Port: 502)
2025-08-03 08:05:32.370 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Reading potenza_prelevata at address 2
2025-08-03 08:09:28.375 ERROR (MainThread) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-08-03 08:09:28.375 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Read Holding Registers modbus_error: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-08-03 08:09:28.375 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] read_modbus_alfa failed: Modbus operation failed: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-08-03 08:09:28.375 ERROR (MainThread) [aiohttp.server] Error handling request from 127.0.0.1
Traceback (most recent call last):
  File "/config/custom_components/sinapsi_alfa/api.py", line 275, in read_holding_registers
    return await self._client.read_holding_registers(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=address, count=count, slave=self._slave_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )  # type: ignore
    ^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 190, in execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request

[... cut ...]
  File "/config/custom_components/sinapsi_alfa/api.py", line 379, in async_get_data
    result = await self.read_modbus_alfa()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/api.py", line 413, in read_modbus_alfa
    raise SinapsiModbusError(f"Failed to read modbus data: {error}") from error
custom_components.sinapsi_alfa.api.SinapsiModbusError: Failed to read modbus data: Modbus operation failed: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request

----

Hope it helps

alexdelprete, 2025-08-03:
There are two issues with your case: getmac isn't able to retrieve the mac address of Alfa. Furthermore, pymodbus doesn't seem able to retrieve data from the device, even though check_port reports that the socket on the device is available for connection. Very weird case. Can you describe your network setup? Is HA running on a separate network vs Alfa? Are using VLANs? Describe your setup a little bit...

Meanwhile, try beta4 in which I tried to address both issues.

JavaJix, 2025-08-04:
Hi Alex,

The server on which is installed HA is on an different VLAN than Alfa that is on the IOT VLAN; but there is a rule on the router to allow routing from the server's VLAN and the IOT one.

It is normal that the Alfa's mac address cannot be read (the ARP table cannot be routed).

But the TCP/IP routing works well, (if i tried a telnet on 502 to Alfa and the server connects). 

I also have modified the HA configuration with modbus and it connects to Alfa and reads the values.

Tried the beta 4, below the logs:

---

2025-08-04 08:05:44.119 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] Test connection to 192.168.10.166:502
2025-08-04 08:05:44.119 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] Creating API Client
2025-08-04 08:05:44.119 DEBUG (MainThread) [custom_components.sinapsi_alfa.config_flow] API Client created: calling get data
2025-08-04 08:05:44.119 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] API Client connect to IP: 192.168.10.166 port: 502 timeout: 5.0
2025-08-04 08:05:44.119 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-04 08:05:44.160 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-04 08:05:44.160 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Inverter ready for Modbus TCP connection
2025-08-04 08:05:44.267 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Modbus TCP Client connected
2025-08-04 08:05:44.267 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-04 08:05:44.365 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-04 08:05:44.365 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 1: SUCCESS
2025-08-04 08:05:44.366 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 1 failed, retrying in 1.1s
2025-08-04 08:05:45.498 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 2 failed, retrying in 2.4s
2025-08-04 08:05:47.898 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 3 failed, retrying in 4.1s
2025-08-04 08:05:51.988 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-04 08:05:52.044 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-04 08:05:52.045 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 4: SUCCESS
2025-08-04 08:05:52.045 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 4 failed, retrying in 8.5s
2025-08-04 08:06:00.582 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 5 failed, retrying in 10.0s
2025-08-04 08:06:10.584 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 6 failed, retrying in 10.0s
2025-08-04 08:06:20.585 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-04 08:06:20.615 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-04 08:06:20.615 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 7: SUCCESS
2025-08-04 08:06:20.615 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 7 failed, retrying in 10.0s
2025-08-04 08:06:30.617 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 8 failed, retrying in 10.0s
2025-08-04 08:06:40.618 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC retrieval attempt 9 failed, retrying in 10.0s
2025-08-04 08:06:50.619 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port: opening socket on 192.168.10.166:502 with a 3.0s timeout.
2025-08-04 08:06:50.743 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Check_Port (SUCCESS): port open on 192.168.10.166:502
2025-08-04 08:06:50.743 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 10: SUCCESS
2025-08-04 08:06:50.743 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC address not found after all attempts
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Using fallback ID: 19216810166_502
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Start Get data (Host: 192.168.10.166 - Port: 502)
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Reading potenza_prelevata at address 2
2025-08-04 08:07:10.749 ERROR (MainThread) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-08-04 08:07:10.749 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Read Holding Registers modbus_error: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-08-04 08:07:10.749 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] read_modbus_alfa failed: Modbus operation failed: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-08-04 08:07:10.749 ERROR (MainThread) [aiohttp.server] Error handling request from fe80::433:94da:8b85:413d
Traceback (most recent call last):
[... cut ...]
  File "/config/custom_components/sinapsi_alfa/api.py", line 383, in async_get_data
    result = await self.read_modbus_alfa()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sinapsi_alfa/api.py", line 417, in read_modbus_alfa
    raise SinapsiModbusError(f"Failed to read modbus data: {error}") from error
custom_components.sinapsi_alfa.api.SinapsiModbusError: Failed to read modbus data: Modbus operation failed: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request

----

Thank you for your support.

alexdelprete, 2025-08-04:
> It is normal that the Alfa's mac address cannot be read (the ARP table cannot be routed).

I know, that is why I asked you about your network config. I imagined there was separation, since getmac usually works in most scenarios.

getmac is used to define a unique_id for the device, since there's no s/n available in modbus registries. As you can see, I implemented a fallback mechanism such that if getmac fails, it uses ip:port to generate the unique_id (it's not really a good practice because if you change IP you will lose data history since the device would be recognised as a different one).

But the real issue is that modbus tcp calls (pymodbus library) aren't able to connect and get data from the device. check_port reports that at low level the stack is able to connect to the port, but pymodbus doesn't.

In the log below is the evidence of what I wrote above: port_check -> getmac fails -> unique_id fallback -> get_data retries for 3 times to retrieve data.

I don't know what else to do sincerely...pymodbus has always worked without issues. Have to think about it...meanwhile, if you can test moving the alfa in the same vlan as HA, just to test if it works, would be a good test to be sure if that is the root cause of the connection issues.
 

```console
2025-08-04 08:06:50.743 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Port check attempt 10: SUCCESS
2025-08-04 08:06:50.743 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] MAC address not found after all attempts
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Using fallback ID: 19216810166_502
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Start Get data (Host: 192.168.10.166 - Port: 502)
2025-08-04 08:06:50.744 DEBUG (MainThread) [custom_components.sinapsi_alfa.api] Reading potenza_prelevata at address 2
2025-08-04 08:07:10.749 ERROR (MainThread) [pymodbus.logging] No response received after 3 retries, continue with next request
```

JavaJix, 2025-08-04:
Hi Alex,

one month ago i have HA installed as OS on a Raspberry pi that uses the same Alfa network and the extention works well.

After that i moved HA to a new server inside docker container on a different VLAN and the extention stops to work.

At the moment i'm not be able to make other test: I assume there is some issue on my network/server configuration.

Many thanks for your support, i will close the issue.

closed 2025-08-04 by JavaJix

alexdelprete, 2025-08-04:
Don't give up...test